moPsy-project/manual-gate-trigger-source #38: ready to rock?

KeLaiFu, 2023-04-04:
I've been watching you work on this project with great interest and noticed you seem to have stopped the regular updates, does this mean the circuit has reached its final form? I have been toying with the same idea, with less success so I am interested in using your schematic as the base of my euro module.

Really appreciate the stuff you're uploading. thanks!

penguineer, 2023-04-04:
Thank you very much for your nice feedback! :smiley: 
You are right, the recent steps in this project didn't leave much of a trace here.

The state of the project:
[The first prototype is done and works.](https://www.patreon.com/posts/first-module-is-80765545) However, while building, I discovered some minor issues with the PCB that need to be worked out.
The circuit itself seems to be working.

So there will be a v1.1 with bugfixes on the PCB.

I am also thinking about a v2 where I drop the voltage switch. This came from a testing environment, but I might solve it differently there and in the performance setup I don't really see as much use for this feature (at least in relation to the effort I made in development and parts to implement this).

penguineer, 2023-04-04:
I am happy to sell you a prototype of v1 for a fair price (covering material and shipping costs), in case you are interested.
As I haven't created build instructions yet, assembling a kit might be a bit of a chore, especially with the current kinks.

Please feel free to get inspiration for your own builds or point out mistakes in my design, :)

penguineer, 2023-04-04:
I am closing this issue and will create new issues with my findings to make it easier to follow up on the progress.

closed 2023-04-04 by penguineer